GyverLibs/EasyHID #3: Ошибка при компиляции для Digispark

korenevskiy, 2021-12-24:
```

In file included from D:\Documents\Arduino\libraries\EasyHID-main\usbdrvasm_includer.S:2:0:
D:\Documents\Arduino\libraries\EasyHID-main\usbdrv/usbdrvasm.S:386:35: fatal error: usbdrvasm165.inc: No such file or directory
 #       include "usbdrvasm165.inc"
                                   ^
compilation terminated.
exit status 1
Ошибка компиляции для платы Digispark (Default - 16.5mhz).

```

.
Еще у Вас в описании есть пример, в этом примере не указан #include.

GyverLibs, 2021-12-24:
Внимательнее к Readme, библиотека работает только на 16 МГц платах. Примеры дополнил, спасибо

GyverLibs, 2021-12-24:
А зачем на дигиспарк, если не секрет? Там своя юсб либа же

GyverLibs, 2021-12-24:
Я добавил файлик под 16.5 МГц, но нужно в usbconfig.h вбить другой порт (B, на тиньке85) и нужный пин. Возможно заведётся

korenevskiy, 2021-12-24:
Я просто зашел на Али, и купил самую дешовую Ардуинку с USB. Прото ее позиционировали как Аналог.
А получается что и не совсем аналог.

korenevskiy, 2021-12-24:
```
#define USB_CFG_IOPORTNAME      B
#define USB_CFG_DMINUS_BIT      4
#define USB_CFG_DPLUS_BIT       3
```
эти параметры?

[image]

korenevskiy, 2021-12-24:
Компиляция прошла успешно.
.
Но виртуальная клава с примером из репозитория текст `Keyboard.println(" Hello world!");  `
Не печатает.

korenevskiy, 2021-12-25:
в конструктор добавил `pinMode(1, OUTPUT);`
А в конец тела функции добавил 
```
  digitalWrite(ledPin, HIGH); 
  delay(200); 
  digitalWrite(ledPin, LOW); 
  delay(200);
```
В результате модифицированный ваш пример 3 раза моргнул светодиодом. и больше ни чего не происходило. Соответственно тексты не пишет.

korenevskiy, 2021-12-25:
Вообще считаю Digispark должен стать более популярным чем Arduino, из-за цены.
Проблема Digispark это отсутствие нативной поддержки у ArduinoIDE. 
Думаю что кто нибудь сделает клон ArduinoIDE со встроенными примерами и драйверами Digispark для пиара своего сайта/форума.

GyverLibs, 2021-12-25:
Абсолютно не согласен. Дигиспарк сейчас сильно подорожал и стоит не сильно дешевле чем та же Nano, а в ней простите возможностей на порядок больше. Если уж говорить о цене, то жалкий Дигиспарк стоит почти столько же, сколько esp8266, где жира ещё больше. 
Arduino ide поддерживает Дигиспарк, достаточно вставить ссылку и установить ядро, как и для всех остальных ардуино-совместимых плат. Делать новую ide и поддерживать её актуальную версию ради предустановленной одной платы (!) Это совсем бред)

korenevskiy, 2021-12-25:
Я Digispark купил за 140р. А Arduino Nano стоит более 270р. А esp8266 нашел за 210р.
С IDE я наверно многого нафантазировал. Изменения наверно излишне.
.
Подскажите я правильные изменения внес в usbconfig.h?
Посоветуйте пожалуйста что может быть причиной не работы Gigispark у меня с Вашим примером работы клавиатуры.

GyverLibs, 2021-12-25:
Раньше 80р он стоил, а есп не подорожала. Если сравнить такие базовые параметры как объем памяти, скорость работы, количество ног... Дигиспарк стоит дороже, чем он должен стоить. 
По usbconfig - подключу автора либы, пусть помогает. 
И всё же, зачем это все если у диги есть своя родная либа для юсб?

korenevskiy, 2021-12-25:
> И всё же, зачем это все если у диги есть своя родная либа для юсб?

@GyverLibs  я уже 3тью либу скачиваю и родную пробывал. кроме родной ни одна не заработала.
Я через либу не могу отправить сигнал Гибернации либо Сна на ПК.
Так как задача в том чтобы при отсутсвии питания в сети выключить ПК. А ИБП сам по себе не имеет интерфейсов в принципе. Он был куплен самый дешовый.
Быть може конечно было бы уже задуматся о смене бесперибойника. Денег жалко, да и уже дело на принцип пошёл.

А у Диджи конечно себестоимость ниже. Но половину суммы съедается на доставку.
Если взять аналогичные товары, например контроллер зарядки для 16850 microUSB. То там видно что 3 устройства стоит 50 рублей, а 6 устройств стоят 70р (или типа в таком смысле.) В любом случае у всех контроллеров доставка предположим одинаковая. А если цена дешевле, то сильно в ущерб самого устройства. С другой стороны, если задачи простые, то зачем переплачивать, разве что только время сэкономить. В одном из роликов сказали что Digi вроде такой же, но везде есть свои особенности, тонкости и пока в каждой тонкости не разберешься хрен что заработает. Но я не знал что могут быть проблемы.

GyverLibs, 2021-12-25:
У программного usb есть ограничения, многие кнопки могут не работать. Лучше взять ту же 32u4 и не мучиться, у нее аппаратная поддержка

korenevskiy, 2021-12-25:
До момента доставки Диджи, я представлял как запускаю Диджи с прошивкой под HID UPS. Чтобы Диджи определялся как реальный UPS. Когда скачал либу под Ардуино, и посмотрел примеры, я просто офигел и понял что не осилю в принципе.. Тем более как бы разобраться с самим конроллером, а потом уже переходить к переделкам либы под свой дижи. И пришел к тому что ограничусь просто прошивкой от клавы с сигналом гибернации.
.
У 32u4 аппаратная поддержка USB? А у нее USB как последовательный порт работает или может работать как что угодно, претворяясь любым устройстом?

GyverLibs, 2021-12-25:
32u4 (Arduino micro) притворяется хидом и чем угодно

korenevskiy, 2021-12-25:
Подключите пожалуйста автора либы.

GyverLibs, 2021-12-25:
Он пока не у компьютера, как придет так придёт

korenevskiy, 2021-12-25:
Подскажите пожалуйста. Если я подключаю две клавы к ПК, то нажав на одной из них CapsLock, то он же загорается и на другой клаве.
А либа эта так же получает сигнал о CapsLock'е?

GyverLibs, 2021-12-25:
Я не знаю, не тестировал библиотеку, её сделал мой коллега

korenevskiy, 2021-12-25:
Только что нашел, [модифицированный](https://sourceforge.net/projects/digistump/files/DigisparkArduino-Win32-1.0.4-May19.zip/download) Arduino IDE со встроенной поддержкой Digispark. 
Правда версия старая. Так как поддержка загрузки либ в IDE появилась недавно. Поэтому наверно ктото модифицировал версию IDE. А возможно производители Digispark это и сделали.

GyverLibs, 2021-12-25:
Зачем? Задача решается в три клика мышкой....

korenevskiy, 2021-12-25:
> Зачем? Задача решается в три клика мышкой....

Ну да теперь, да.  Но чтобы это сделать надо инструкцию посмотреть. Я вот посмотрел кучу инструкций где это рассказывалось. кое в чем разобрался. А без инструкций не чего не получится. Я работаю UX разработчиком. 
Стараемся делать софт интуитивно понятный. Пример хорошей адаптивности. Например как в онлайн сбербанке  в поле пароля программа удаляла пробелы за звездочками. Или у них пароли были одинаково восприимчивы к обоим регистрам. Фишка в том что на клавиатуре написаны буквы в большом регистре, но когда человек пишет, то почему то буквы пишутся в маленьком регистре. А за звездочками вообще не видно. Это для программистов и компьютерных аналитиков и дизайнеров все очевидно, а для всех остальных пенсионеров и просто других людей некоторые вещи вообще не очевидны и не понятны.
Я вот после первой прошивки Диджи 2 часа потратил на выяснение почему Диджи перестал прошиваться. Оказывается во всех инструкциях написано и звучит один в один фраза "Ардуино Digispark не нужно заранее подключать к компьютеру." или
[image]
Как это нужно понимать? слово "не нужно" воспринимается как не обязательно (типа можно делать а можно и не делать). 
Просто иногда оцениваю простоту экслуатации интерфейсов и иногда надуманно преувеличиваю проблему.

GyverLibs, 2021-12-25:
Каких кучу инструкций, зачем?? На официальной Вики дигистамп всё написано по пунктам

korenevskiy, 2021-12-25:
На офицальной Вики все на английском, русского нет.
Я пользовался инструкциями из яндекса и из ютюба.

korenevskiy, 2021-12-25:
Жалко что ArduinoIDE v2 не поддерживает Digi.

GyverLibs, 2021-12-25:
Инструкции на русском часто нерабочие, или работают только у автора :)

korenevskiy, 2021-12-26:
> Инструкции на русском часто нерабочие, или работают только у автора :)

Вы поддержку Диджи добавляли, это Вы добавили usbdrvasm165.inc. А этот файлик является копией usbdrvasm16.inc?

korenevskiy, 2021-12-26:
А Вы пробовали [Platformio](https://youtu.be/OGCyKncOyNU)?

GyverLibs, 2021-12-26:
> является копией usbdrvasm16.inc?
Не знаю, автор либы скинул

> Вы пробовали Platformio?
Да, ужасно неудобная и кривая IDE

korenevskiy, 2021-12-27:
Дешёвый аналог на STM32 нашёл. Стоит 160р. И работает на 72МГц и памяти всех видов больше .

GyverLibs, 2021-12-27:
Осталось научиться с ним работать)

korenevskiy, 2021-12-27:
Нашел скетч для Arduino, который превращает его в ИБП UPS USB. Совместим со встроенными драйверами Windows и Linux.
[https://create.arduino.cc/projecthub/abratchik/hid-compliant-ups-with-arduino-75198c](https://create.arduino.cc/projecthub/abratchik/hid-compliant-ups-with-arduino-75198c)
К сожалению  опять таки Digi выдает ошибку при компиляции.

GyverLibs, 2021-12-27:
Очевидно, библиотека под 32U4

Kiprus, 2022-01-19:
> ```
> #define USB_CFG_IOPORTNAME      B
> #define USB_CFG_DMINUS_BIT      4
> #define USB_CFG_DPLUS_BIT       3
> ```
> 
> эти параметры?
> 
> [image]

Как минимум из того что я могу сказать, в библиотеке используется INT0 прерывание, а вы используете пины B3 и B4 которые не являются прерываниями, вот оно и не работает.

По зависанию тоже есть предположение, библиотека инициализацирует INT0 прерывание на спад а из-за того что пин INT0 весит в воздухе и ловит помехи прерывание срабатывает каждую секунду чем не даёт выполнятся основному коду

Kiprus, 2022-01-19:
> > Как минимум из того что я могу сказать, в библиотеке используется INT0 прерывание, а вы используете пины B3 и B4 которые не являются прерываниями, вот оно и не работает.
> > По зависанию тоже есть предположение, библиотека инициализирует INT0 прерывание на спад а из-за того что пин INT0 весит в воздухе и ловит помехи прерывание срабатывает каждую секунду чем не даёт выполнятся основному коду
> 
> Посоветуйте пожалуйста, какие именно параметры можно поменять. Примерное наименование параметров в либе скажите, буду очень признателен. INT0 у меня подключен к резистору на плюс. Как его програмно отключить? Метод `pinMode();` отключит прерывание на нем? PS интересно то что изначально припаяв на P5 сигнал от внешнего БП о наличии питания в сети. Диджи постоянно глючил. Потом понял что P5 работает как сброс. Думается что этот пин переназначить.

Без прерывания либа не будет работать, прерывание сделанно для получения данных по usb так что без него hid работать не будет

korenevskiy, 2022-01-19:
> > > Как минимум из того что я могу сказать, в библиотеке используется INT0 прерывание, а вы используете пины B3 и B4 которые не являются прерываниями, вот оно и не работает.
> > > По зависанию тоже есть предположение, библиотека инициализирует INT0 прерывание на спад а из-за того что пин INT0 весит в воздухе и ловит помехи прерывание срабатывает каждую секунду чем не даёт выполнятся основному коду
> > 
> > Посоветуйте пожалуйста, какие именно параметры можно поменять. Примерное наименование параметров в либе скажите, буду очень признателен. INT0 у меня подключен к резистору на плюс. Как его програмно отключить? Метод `pinMode();` отключит прерывание на нем? PS интересно то что изначально припаяв на P5 сигнал от внешнего БП о наличии питания в сети. Диджи постоянно глючил. Потом понял что P5 работает как сброс. Думается что этот пин переназначить.
> 
> Без прерывания либа не будет работать, прерывание сделанно для получения данных по usb так что без него hid работать не будет
> 
> Из того что могу посоветовать это перекинуть Data- на PB2. И ещё важно что бы pull-up резистор(1.5k/2.2K) подключались программно (у дигиспарк припаян резистор между d- и 5V из-за чего даже без запуска библиотеки хост(ПК/ноут и т.д) начинает опрашивать Hid, а библиотека не отвечает)
> 
> Но что бы не парится этим всем лучше использовать либу которая создана специально для digispark

Конечно, я бы с радостью бы ее использовал, но там нет поддержки клавиши Сна. Конечно там есть печать символа по коду, но я не нашёл код клавиши сна. Я пробовал , не получалось. Быть может я пытался печатать правильный код, но делал это не в той кодировке.

closed 2023-05-18 by GyverLibs