GyverLibs/EasyHID #4: Unknown device

Masim228, 2021-12-31:
Не понимаю почему пишет про не известный девайс, пробовал и на Uno, и на Nano, делал по инструкции, библиотеки позакидал.

GyverLibs, 2022-01-01:
Вроде у всех пока работало, не знаю

AAKOTb, 2022-02-16:
Подтверждаю, та же ошибка, пробовал на китайской и оригинальной УНО. Подключение делал по первой схеме со стабилитронами на 3.6, использовал 1N4729A. Проверял на разных компьютерах, везде "Устройство ЮСБ не опознано"

wheelround, 2022-08-12:
Добрый день. Нашли решение этой проблемы или нет? А то все схемы собрал, и со всеми схемами такая муть :с
Система пишет, что неопознанное устройство (сбой дескриптора устройства)

Teresk1, 2022-08-19:
На андроиде прекрасно работает на ноуте виндовс 7 проц i5  работает на ПК i5  виндовс 10 работает, а на ПК 775 сокет xeon e5440 и e8400 виндовс 10 Запуск этого устройства невозможен. (Код 10) .