GyverLibs/EasyHID #6: Ошибка дескритора. HID не отправляет Handshake

Kiprus, 2022-01-18:
В моём HID устройстве есть нужда работать от аккумулятора и отключаться от usb при это оставаясь в работе.
Виходит так что при первом подключении к usb HID определяется а после переподключения уже появляется ошибка дескриптора, с помощью логического анализатора увидел что после переподключения HID не отправляет пакет Handshake.
Вот кусок кода: https://pastebin.com/PdJRAhVd

Использую микроконтролер Atmega8 16mhz и среду разработки PlatformIO

Первое подключение:
[image]

Второе подключение:
[image]

Kiprus, 2022-01-19:
for(int i; i<10; i++;){
Я идиот(ну или компилятор)
}

Проблема оказалась в цикле for.
Он у меня был написан:
for(int i; i<10; i++;){код}
и при повторном вызове переменная i была больше или ровно 10 из-за чего код HID.tick() не выполнялся, вот или я не правильно понимал вызов функции но в моём понимании int i; должен сбрасывать функцию в 0, но он этого не делал.
И ещё одна странность, не смотря на то что HID.tick() не вызывался клавиша эмулировалась, что и ввела меня в заблуждение и двухдневное мучение и себя и вас.
Спасибо всем кто пытался помочь.

closed 2022-01-19 by Kiprus

GyverLibs, 2022-01-19:
Локальные переменные при создании имеют практически случайное значение, их обязательно нужно инициализировать если например нужен 0 для счётчика. Это очень базовые вещи)