GyverLibs/EasyHID #9: Ошибка #define EASYHID_SOFT_DETACH

aruytehno, 2022-04-08:
С версией библиотеки v1.0 данной проблемы не наблюдается
```
In file included from /Users/aruytehno/Documents/Arduino/libraries/EasyHID/HIDPrivate.h:8:0,
                 from /Users/aruytehno/Documents/Arduino/libraries/EasyHID/EasyHID.h:23,
                 from /Users/aruytehno/Documents/Arduino/sketch_apr08a/sketch_apr08a.ino:13:
/Users/aruytehno/Documents/Arduino/libraries/EasyHID/usbconfig.h:29:24: error: missing expression between '(' and ')'
 #if(EASYHID_SOFT_DETACH)     // PULLUP по желанию
                        ^
Несколько библиотек найдено для "EasyHID.h"
 Используется: /Users/aruytehno/Documents/Arduino/libraries/EasyHID
Не используется: /Users/aruytehno/Documents/Arduino/libraries/EasyHID 2
exit status 1
Ошибка компиляции для платы Arduino Uno.
```

Nich1con, 2022-04-08:
1. Удалите уже старую версию библиотеки
2. При обьявлении  #define EASYHID_SOFT_DETACH скобки НЕ нужны (специально проверил).

Nich1con, 2022-04-08:
> С версией библиотеки v1.0 данной проблемы не наблюдается
>
> ```
> In file included from /Users/aruytehno/Documents/Arduino/libraries/EasyHID/HIDPrivate.h:8:0,
>                  from [...]

1. Удалите уже старую версию библиотеки
2. При обьявлении #define EASYHID_SOFT_DETACH скобки НЕ нужны.
3. В какой среде компилируете код?

GyverLibs, 2022-04-08:
Пример кода в студию? Что вызывает ошибку?

aruytehno, 2022-04-08:
EASYHID_SOFT_DETACH - увидел, реализовано только во второй версии, поэтому с первой версией библиотекии не ругалось. 

Ругается на пример с описания

/*
   Пример эмуляции клавиатуры EasyHID
   Можно нажимать, отжимать и кликать одновременно до 5ти клавиш
   Нажатые клавиши буфферизируются, клавиши модификаторы (ALT,CTRL...) так же работают
   Список всех доступных клавиш см. в файле Codekeys.h

   Если используется активная подтяжка при помощи пина,
   обьявляем "EASYHID_SOFT_DETACH" ПЕРЕД подключением EasyHID.h
   (Раскоментируйте строку ниже)
*/
```
#define EASYHID_SOFT_DETACH

#include <EasyHID.h>

void setup() {
  HID.begin();                          // Инициализация USB
}

void loop() {
  static uint32_t timer = millis();     // Миллис - таймер (НЕ DELAY!!!)
  if (millis() - timer >= 1000) {       // Каждые 1000 мс
    timer = millis();

    Keyboard.press(KEY_1);              // Зажимаем клавишу '1'
    Keyboard.releaseAll();              // Отпускаем все клавиши
    Keyboard.click(KEY_SPACE);          // Кликаем пробел (press + releaseAll)
    Keyboard.write('-');                // Выводим тире
    Keyboard.println(" Hello world!");  // Печать в println
  }

  HID.tick();   // Вызываем тикер не реже чем каждые 10 мс!
}
```

aruytehno, 2022-04-08:
Что если не отключать usb при заливке прошивки?

GyverLibs, 2022-04-08:
Пофикшено

closed 2022-04-08 by GyverLibs