GyverLibs/EasyHID #11: Пин Data- на ATtiny88 (плата MH-ET)

Shrek27, 2022-07-03:
Платка MH-ET Tiny не хотела определяться как HID-устройство. В интернетах нарыл что PD1 висит на пине 1, переткнул и все заработало.

GyverLibs, 2022-07-03:
Так, а что где конкретно перекинул?

Shrek27, 2022-07-08:
Линию D- с пина 0 на пин 1.

Nich1con, 2022-07-15:
> Платка MH-ET Tiny не хотела определяться как HID-устройство. В интернетах нарыл что PD1 висит на пине 1, переткнул и все заработало.

В текущей версии либы (2.2.1) платка HM-88 корректно определяется "из коробки" (с пином PD0 на D-). Для платы Attiny88 с загрузчиком micronucleous пины usb захардкожены в коде загрузчика, и мы сомневаемся что существует две версии платы с разной распайкой D-. Попробуйте полностью удалить либу из папки libraries и установить последнюю версию.

Shrek27, 2022-07-17:
[image]
В файле usbconfig.h четко прописаны пины 1 и 2 для плат ATtiny48 и ATtiny88 (строчки 11 и 12) и у меня соответсвенно так и заработало. А в описании библиотеки (https://github.com/GyverLibs/EasyHID) для MH-ET указан пин 0 для линии D-. Библиотека работает прекрасно, просто кто-то из новичков (типа меня =) ) может поначалу запутаться, если не полезет глубже первой страницы.

GyverLibs, 2022-07-17:
Тебе автор пишет, что у него на такой же плате всё работает из коробки. Нужно разобраться в чем дело, а не лезть в исходник и за новичков переживать))

GyverLibs, 2022-07-17:
Так погоди, у этой платы юсб распаян на штекер, никакие резисторы не нужны . Просто микро юсб подключи и всё....

Shrek27, 2022-07-17:
> Так погоди, у этой платы юсб распаян на штекер, никакие резисторы не нужны . Просто микро юсб подключи и всё....

О_О пойду тестить, это очень-очень... очень сильно упростит мне задачу!

closed 2022-07-19 by Shrek27