GyverLibs/EasyHID #12: Не собирается под ATmega8 и ядром miniCore.

DmitryTatarinov, 2022-07-06:
Вроде в описании указана поддержка ATmega8. Но по факту ее нет. 
Попробовал добавить ее в usbconfig.h в секцию для голых камней. Начало собираться, но только при использовании внешнего кварца. Если выбрать внутренний на 8МГц, то при компиляции вот такая ошибка:

H:\\arduinoPortable\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega8 -DF_CPU=8000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega8 -DARDUINO_ARCH_AVR "-IH:\\arduinoPortable\\portable\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IH:\\arduinoPortable\\portable\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IH:\\arduinoPortable\\libraries\\EasyHID-main" "H:\\arduinoPortable\\libraries\\EasyHID-main\\usbdrvasm_includer.S" -o ..\\AppData\\Local\\Temp\\arduino_build_567725\\libraries\\EasyHID-main\\usbdrvasm_includer.S.o"
In file included from H:\arduinoPortable\libraries\EasyHID-main\usbdrvasm_includer.S:2:0:
H:\arduinoPortable\libraries\EasyHID-main\usbdrv/usbdrvasm.S:390:9: error: #error "USB_CFG_CLOCK_KHZ is not one of the supported non-crc-rates!"
 #       error "USB_CFG_CLOCK_KHZ is not one of the supported non-crc-rates!"
         ^~~~~
Используем библиотеку EasyHID-main в папке: H:\arduinoPortable\libraries\EasyHID-main (legacy)
exit status 1
Ошибка компиляции для платы ATmega8.

Возможна ли работа данного камня от внутреннго кварца с этой либой?

GyverLibs, 2022-07-06:
Через неделю коллега вернётся который работал над либой, даст ответ

GyverLibs, 2022-07-15:
Пофиксили

closed 2022-07-15 by GyverLibs

DmitryTatarinov, 2022-07-17:
При использовании внутреннего кварца для тактирования, ошибка компиляции никуда не делась. См. первый пост.

GyverLibs, 2022-07-17:
библиотека в принципе не работает от 8 МГц, об этом даже в ошибке написано