GyverLibs/EasyHID #14: Ctrl+Alt+Tab = Переход в гибернацию

TRRAXT, 2022-09-10:
При нажатии на энкодер и прокрутке должно срабатывать нажатие Alt+Tab, но почему-то происходит отправка девайса в гибернацию(комп/ноут, на Android просто блокировка). Может я делаю что-то не так?

`void loop() {
  if (enc.tick()) {
    if (enc.isClick())  Keyboard.clickMultimediaKey(KEY_PLAYPAUSE);           // Клик по кнопке
    else if (enc.isLeft())  Keyboard.clickMultimediaKey(KEY_VOL_UP);          // Вращение вправо
    else if (enc.isRight())   Keyboard.clickMultimediaKey(KEY_VOL_DOWN);      // Вращение влево
    else if (enc.isLeftH()){                                                  // Вращение вправо с нажатием
      Keyboard.clickSystemKey(KEY_LEFT_ALT);
      Keyboard.click(KEY_TAB);
    }
    else if (enc.isRightH()){                                                 // Вращение влево с нажатием
      Keyboard.clickSystemKey(KEY_LEFT_ALT);
      Keyboard.clickSystemKey(KEY_LEFT_SHIFT);
      Keyboard.click(KEY_TAB);
    }
  }`

ATTiny88(MH-Tiny), не знаю какая версия библиотеки(скачал с пункта "Установка"), Arduino IDE 1.8.19, стандартные примеры работают хорошо.

GyverLibs, 2022-09-10:
привет, исправил документацию, системные клавиши это только клавиши для сна и выключения. Все остальные обрабатываются через клик

closed 2022-09-10 by GyverLibs

TRRAXT, 2022-09-10:
т.е никакого аль+таба?