GyverLibs/EasyHID #21: Не определяется виндой

swaftrade, 2023-04-13:
Win 11
IDE v2.0.4
библиотека скачана сегодня, версию не могу найти...
UNO 16Мгц 328p
провод 70 см
собрана схема со стабилитронами

винда никак не реагирует на подключение USB

GyverLibs, 2023-04-13:
на всякий случай можно попробовать прошить из ide первой версии

swaftrade, 2023-04-13:
> на всякий случай можно попробовать прошить из ide первой версии

1.8.19 ничего не изменила. Есть идеи?

GyverLibs, 2023-04-13:
Неправильная/кривая пайка

swaftrade, 2023-04-14:
> Неправильная/кривая пайка

Утро вечера мудренее...
перепроверил 100500 раз. Пробовал Nano 328p 16MHz. Другой кабель USB. Осциллограф на портах показывает "смертельную" :) линию.
[image]

KralichkinV, 2023-06-03:
> ###
> [Apr 14, 2023](#issuecomment-1508015207)

у вас в D5 вместо питания резистор заведен, и поэтому usb не видит low speed

Sergio-tix, 2023-07-17:
> у вас в D5 вместо питания резистор заведен, и поэтому usb не видит low speed

Какое питание?
Вот схема автора:
[image]
У [swaftrade](https://github.com/swaftrade) всё по схеме собрано.

P.S. Тоже не работает ни в какую.

GyverLibs, 2023-07-17:
Собирайте по актуальным схемам со страницы библиотеки. Если не работает - часто бывает виноват китайский кварц на ардуине.

swaftrade, 2023-07-18:
> > у вас в D5 вместо питания резистор заведен, и поэтому usb не видит low speed
> 
> Какое питание? Вот схема автора: [image] У [swaftrade](https://github.com/swaftrade) всё по схеме собрано.
> 
> P.S. Тоже не работает ни в какую.

Я плюнул на это дело. Поставил Leonardo. Всё работает

Sergio-tix, 2023-07-18:
> Я плюнул на это дело.

Я полазил по инетам, эта тема тянется с 2013 года, и способ этот 50/50 в лучшем случае. Работает не стабильно и на грани. И не этим автором едины, как минимум три библиотеки нашёл. Сразу скажу, не взлетела ни одна. А от библиотеки зависимость есть. Например с TrinketHid коннект прошёл, но отвалилось через минуту. А с текущей бибиотекой на том же железе не алё.  
В общем да, пришлось срочняком в офлайне взять Attiny85 Digispark, и вот там всё заработало.
P.S. При загрузке видеорегистратор включается по умолчанию в режиме без звука. Вот и пришлось реализовать эмуляцию мыши, чтоб через минуту после подачи питания курсор сам щёлкнул по пиктограмме динамика.

GyverLibs, 2023-07-18:
Эта библиотека работает на базе VUSB, как и многие другие (почти все) подобные проекты по эмуляции HID на AVRках. Работает действительно на грани возможностей МК и очень чувствительна к кварцу, если китайская плата неудачная - не будет работать ни одна другая библиотека на базе VUSB. У нас таких случаев не было, всё работает как часы

yojig666, 2024-04-12:
у меня тоже на arduino uno не работает. пробовал DFRduino Duemilanove, тоже самое. пойду копать в сторону 32u4, с ним тач когда-то собирал и клавомышь эмулировал