GyverLibs/EasyHID #22: Вставка текста на русском языке

kraa965, 2023-05-16:
Привет, подскажите, как можно реализовать вставку русского текста, в видео выводился английский?

GyverLibs, 2023-05-18:
Alt+Shift?

kraa965, 2023-05-18:
я имею ввиду через print("Привет"), вместо привет, он выдает символы и другие буквы

GyverLibs, 2023-05-18:
Ну он и не должен выдавать русские, это ведь клавиатура. Нужно переключить язык и писать английскими, которые соответствуют русским

kraa965, 2023-05-18:
> Ну он и не должен выдавать русские, это ведь клавиатура. Нужно переключить язык и писать английскими, которые соответствуют русским

Получается, надо написать print("Ghbdtn"), переключить клаву и будет все по-русски. 
А если надо передать текст из веб, ввожу в поле русский текст, а выдаются просто символы, как бы в таком случае?

GyverLibs, 2023-05-18:
Надо писать свой парсер, который будет вытаскивать русские буквы из текста, переводить их в англ согласно qwerty раскладке, формировать строку и отправлять в принт. А если текст смешанный русско-англ - переключать язык, выдерживать паузу, но опять же обратной связи по языку нет, можно сбиться

GyverLibs, 2023-05-18:
Можно посмотреть как парсит библиотека gyveroled, там все русские буквы обрабатываются

kraa965, 2023-05-18:
> Можно посмотреть как парсит библиотека gyveroled, там все русские буквы обрабатываются

Есть же еще библиотека GParser, она может помочь?

kraa965, 2023-05-18:
> Gparser разбивает текст на куски по заданному символу-разделителю

Видимо долблюсь в глаза, не нашел в gyveroled парсер, подскажите где она?

GyverLibs, 2023-05-18:
Со строки 275 по 285
https://github.com/GyverLibs/GyverOLED/blob/main/src/GyverOLED.h

И дальше внизу функция getFont разбирает в таблицу

GyverLibs, 2023-05-18:
Я могу закодить, но через несколько дней

kraa965, 2023-05-18:
> Я могу закодить, но через несколько дней

буду очень сильно признателен

GyverLibs, 2023-05-24:
Вот так можно преобразовать русскую строку в QWERTY
```cpp
const char _qwerty_ru[] PROGMEM = "F<DULT:PBQRKVYJGHCNEA{WXIO}SM\">Zf,dult;pbqrkvyjghcnea[wxio]sm'.z~`";

String ru_to_qw(const String& ru) {
  String qw;
  uint8_t prev = 0;
  for (int i = 0; i < ru.length(); i++) {
    uint8_t cur = ru[i];
    if (cur > 127) {
      uint8_t thiscur = cur;
      if (cur > 191) cur = 0;
      else if (prev == 209 && cur == 145) cur = 193;   // ё
      else if (prev == 208 && cur == 129) cur = 192;   // Ё
      prev = thiscur;
    }
    if (!cur) continue;
    if (cur <= 127) ; // skip eng
    else if (cur <= 143) cur -= 80;
    else if (cur <= 191) cur -= 144;
    else cur -= 128;
    qw += (char)pgm_read_byte(&_qwerty_ru[cur]);
  }
  return qw;
}

void setup() {
  Serial.begin(115200);
  Serial.println(ru_to_qw("Привет")); // Ghbdtn
}

void loop() {
}

```

kraa965, 2023-05-24:
> Вот так можно преобразовать русскую строку в QWERTY
> ```cpp
> const char _qwerty_ru[] PROGMEM = "F<DULT:PBQRKVYJGHCNEA{WXIO}SM\">Zf,dult;pbqrkvyjghcnea[wxio]sm'.z~`";
> 
> String ru_to_qw(const String& ru) {
>   String qw;
>   uint8_t prev = 0;
>   for (int i = 0; i < ru.length(); i++) {
>     uint8_t cur = ru[i];
>     if (cur > 127) {
>       uint8_t thiscur = cur;
>       if (cur > 191) cur = 0;
>       else if (prev == 209 && cur == 145) cur = 193;   // ё
>       else if (prev == 208 && cur == 129) cur = 192;   // Ё
>       prev = thiscur;
>     }
>     if (!cur) continue;
>     
>     if (cur <= 127) {
>       qw += (char)cur;
>       continue;
>     }
>     else if (cur <= 143) cur -= 80;
>     else if (cur <= 191) cur -= 144;
>     else cur -= 128;
>     qw += (char)pgm_read_byte(&_qwerty_ru[cur]);
>   }
>   return qw;
> }
> 
> void setup() {
>   Serial.begin(115200);
>   Serial.println(ru_to_qw("Привет")); // Ghbdtn
> }
> 
> void loop() {
> }
> 
> ```

Спасибо)
Вечером попробую

GyverLibs, 2023-06-18:
> она у меня не работает

она и не должна работать. Кириллица занимает больше одного байта, писать `'б'` - уже некорректно, буква потеряна. Компилятор должен был об этом предупредить

kraa965, 2023-06-20:
> Вот так можно преобразовать русскую строку в QWERTY
> ```cpp
> const char _qwerty_ru[] PROGMEM = "F<DULT:PBQRKVYJGHCNEA{WXIO}SM\">Zf,dult;pbqrkvyjghcnea[wxio]sm'.z~`";
>
> String ru_to_qw(const String& ru) [...]

Можно этот код под char* адаптировать?

GyverLibs, 2023-06-20:
Естественно

kraa965, 2023-06-20:
> Естественно

Подскажите как?

GyverLibs, 2023-06-20:
Ну например так
```cpp
const char _qwerty_ru[] PROGMEM = "F<DULT:PBQRKVYJGHCNEA{WXIO}SM\">Zf,dult;pbqrkvyjghcnea[wxio]sm'.z~`";

uint16_t ru_to_qw(const char* ru, char* qw) {
  uint16_t len = strlen(ru);
  uint16_t idx = 0;
  uint8_t prev = 0;
  for (int i = 0; i < len; i++) {
    uint8_t cur = ru[i];
    if (cur > 127) {
      uint8_t thiscur = cur;
      if (cur > 191) cur = 0;
      else if (prev == 209 && cur == 145) cur = 193;   // ё
      else if (prev == 208 && cur == 129) cur = 192;   // Ё
      prev = thiscur;
    }
    if (!cur) continue;
    
    if (cur <= 127) {
      qw[idx++] = (char)cur;
      continue;
    }
    else if (cur <= 143) cur -= 80;
    else if (cur <= 191) cur -= 144;
    else cur -= 128;
    qw[idx++] = (char)pgm_read_byte(&_qwerty_ru[cur]);
  }
  qw[idx] = 0;
  return idx;
}

void setup() {
  Serial.begin(115200);

  char* ru = "Привет";
  char qw[strlen(ru)];
  ru_to_qw(ru, qw);

  Serial.println(qw);  // Ghbdtn
}

void loop() {
}
```

kraa965, 2023-06-20:
> Ну например так
> ```cpp
> const char _qwerty_ru[] PROGMEM = "F<DULT:PBQRKVYJGHCNEA{WXIO}SM\">Zf,dult;pbqrkvyjghcnea[wxio]sm'.z~`";
>
> uint16_t ru_to_qw(const char* ru, char* qw) {
>   uint16_t len = [...]

Спасибо)

closed 2024-01-25 by GyverLibs