GyverLibs/EasyHID #35: Сбой запроса дескриптора USB-устройства

nTURBOn, 2023-11-19:
После подключения к компьютеру пишет 
`Система Windows остановила это устройство, так как оно сообщило о возникновении неполадок. (Код 43)

Сбой запроса дескриптора USB-устройства.`

Собрано на Arduino nano, windows 10. в коде только подключение библиотеки и HID.tick();

progzone122, 2024-05-26:
Получилось решить проблему? Столкнулся с тем же на arduino uno.
Пишет: Неизвестное USB-устройство (сбой запроса дескриптора устройства) Код 43

nTURBOn, 2024-05-26:
К сожалению, нет, просто заказал arduino pro micro на озоне (так было дешевле) и очень сильно помог [гайд](https://www.thingiverse.com/thing:4627779) там и видео на ютубе и мануал есть. Сделал свою 3д модель по основам этого гайда и чуть изменил прошивку со схемой (просто подключал кнопки "сеткой" + добавил энкодер с функцией нажатия для изменения громкости и остановки видео). Я вообще делал что-то на подобе стрим дека, поэтому использовал программу CoyoteMIDI что бы преобразовать MIDI сигналы от arduino в сочетания клавиш для обс, дискорда и т. д. Получился довольно удобный инструмент, до сих пор пользуюсь, проблем нет.